openwisp/openwisp-monitoring #317: [bug] Wifi Clients getting overwritten if datetime if passed explictly

devkapilbansal, 2021-08-12:
Found this bug while working on #316 

Currently [datetime is added to metric just before writing it to timeseries database](https://github.com/openwisp/openwisp-monitoring/blob/d2994990a8171017cca636747b75dd2c3f4a6386/openwisp_monitoring/monitoring/base/models.py#L221).

But if we pass datetime, say from API, or [here](https://github.com/openwisp/openwisp-monitoring/blob/d2994990a8171017cca636747b75dd2c3f4a6386/openwisp_monitoring/device/api/views.py#L243) before for loop, then same timestamp will be passed for all clients.

This may be resulting in overwriting the clients.

** How to Reproduce this Error**
- Modify [this line](https://github.com/openwisp/openwisp-monitoring/blob/d2994990a8171017cca636747b75dd2c3f4a6386/openwisp_monitoring/device/api/views.py#L246) as `metric.write(client['mac'], time=now())` locally.
- Run tests and see the error.

The tests fails [here](https://github.com/openwisp/openwisp-monitoring/blob/d2994990a8171017cca636747b75dd2c3f4a6386/openwisp_monitoring/device/tests/__init__.py#L76). No. of points received will be 1, that is not equal to wireless clients available in tests. ([2 for now](https://github.com/openwisp/openwisp-monitoring/blob/d2994990a8171017cca636747b75dd2c3f4a6386/openwisp_monitoring/device/tests/__init__.py#L221-L252))

nemesisdesign, 2021-08-20:
@devkapilbansal the solution is to add a microsecond difference in each client:

```python
from datetime import timedelta

# ....
client_time = self.time
for client in clients:
    if 'mac' not in client:
        continue
    metric.write(client['mac'], time=client_time)
    client_time += timedelta(microseconds=1)
```

devkapilbansal, 2021-08-20:
@nemesisdesign although this will work but I think ideally metric shouldn't overwrite clients if same datetime is passed.

nemesisdesign, 2021-08-20:
> @nemesisdesign although this will work but I think ideally metric shouldn't overwrite clients if same datetime is passed.

@devkapilbansal it's not something we can control: https://github.com/influxdata/influxdb/issues/4150#issuecomment-141186561

nemesisdesign, 2021-09-19:
Has been fixed in https://github.com/openwisp/openwisp-monitoring/pull/316.

commit 1c3f3075 (2021-08-22): [fix] Fix wifi clients getting overwritten if explicit datetime is passes #317

Closes #317
closed 2021-09-19 by nemesisdesign