openwisp/openwisp-monitoring #322: [deps] Add support for django 3.2 and django 4.0 alpha

codesankalp, 2021-09-27:
et's make sure this module can run with Django 3.2 and Django 4.0 alpha.

ashutosh887, 2021-11-29:
I would like to take this up @codesankalp

Aryamanz29, 2021-12-22:
@codesankalp How to proceed with this issue? Could you please guide

codesankalp, 2021-12-22:
> @codesankalp How to proceed with this issue? Could you please guide

Try testing openwisp-monitoring for different versions of Django as mentioned in the issue heading and if some issue arises while testing, fix them.
 See https://github.com/openwisp/openwisp-utils/pull/258 for reference.

pandafy, 2021-12-30:
Hey @Aryamanz29! Are you still working on this?

Aryamanz29, 2021-12-30:
> Hey @Aryamanz29! Are you still working on this? 

 Yes @pandafy , I'll be raise PR soon.

commit 28abf656 (2022-01-10): [deps] Added support for Django 3.2 and Django 4.0 #322

- Bumped django-nested-admin~=3.4.0
- Pinned test dependencies

Closes #322
commit 175d8eff (2022-01-11): [deps] Added support for Django 3.2 and Django 4.0 #322

- Bumped django-nested-admin~=3.4.0
- Pinned test dependencies

Closes #322
closed 2022-01-11 by nemesisdesign