openwisp/openwisp-monitoring #348: Applying check.0003_create_ping fails

alangecker, 2022-03-18:
after upgrading & enabling monitoring the migrations fail now with following error:

`django.core.exceptions.FieldError: Cannot resolve keyword 'check_type' into field. Choices are: active, check, content_type, 
content_type_id, created, description, id, modified, name, object_id, params`

since I'm neither familiar with openwisp nor django, I struggle a bit debugging it. Does anyone have some hints? 😃 

-----

### Full stack trace
```python
Traceback (most recent call last):
  File "/opt/openwisp/manage.py", line 9, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.10/site-packages/django/core/management/__init__.py", line 446, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.10/site-packages/django/core/management/__init__.py", line 440, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/usr/local/lib/python3.10/site-packages/django/core/management/base.py", line 414, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/usr/local/lib/python3.10/site-packages/django/core/management/base.py", line 460, in execute
    output = self.handle(*args, **options)
  File "/usr/local/lib/python3.10/site-packages/django/core/management/base.py", line 98, in wrapped
    res = handle_func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/core/management/commands/migrate.py", line 290, in handle
    post_migrate_state = executor.migrate(
  File "/usr/local/lib/python3.10/site-packages/django/db/migrations/executor.py", line 131, in migrate
    state = self._migrate_all_forwards(
  File "/usr/local/lib/python3.10/site-packages/django/db/migrations/executor.py", line 163, in _migrate_all_forwards
    state = self.apply_migration(
  File "/usr/local/lib/python3.10/site-packages/django/db/migrations/executor.py", line 248, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/usr/local/lib/python3.10/site-packages/django/db/migrations/migration.py", line 131, in apply
    operation.database_forwards(
  File "/usr/local/lib/python3.10/site-packages/django/db/migrations/operations/special.py", line 193, in database_forwards
    self.code(from_state.apps, schema_editor)
  File "/usr/local/lib/python3.10/site-packages/openwisp_monitoring/check/migrations/0003_create_ping.py", line 13, in create_device_ping
    auto_create_ping(
  File "/usr/local/lib/python3.10/site-packages/celery/local.py", line 188, in __call__
    return self._get_current_object()(*a, **kw)
  File "/usr/local/lib/python3.10/site-packages/celery/app/task.py", line 392, in __call__
    return self.run(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/openwisp_monitoring/check/tasks.py", line 63, in auto_create_ping
    has_check = Check.objects.filter(
  File "/usr/local/lib/python3.10/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 1071, in filter
    return self._filter_or_exclude(False, args, kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 1089, in _filter_or_exclude
    clone._filter_or_exclude_inplace(negate, args, kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 1096, in _filter_or_exclude_inplace
    self._query.add_q(Q(*args, **kwargs))
  File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1466, in add_q
    clause, _ = self._add_q(q_object, self.used_aliases)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1496, in _add_q
    child_clause, needed_inner = self.build_filter(
  File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1342, in build_filter
    lookups, parts, reffed_expression = self.solve_lookup_type(arg)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1156, in solve_lookup_type
    _, field, _, lookup_parts = self.names_to_path(lookup_splitted, self.get_meta())
  File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1640, in names_to_path
    raise FieldError(

```

kitchung, 2022-05-13:
FYI, I have the same error; still researching.

kitchung, 2022-05-13:
I suspect this may be the cause:
https://github.com/openwisp/openwisp-monitoring/commit/dedc4a913004e019a74eef6fcb64647ecdf31114

nemesisdesign, 2022-05-13:
The most important part of the stack trace is not shown, we never encountered this issue, so not being able to reproduce it, you have to help us understand what's going on otherwise there's no way we can help.

kitchung, 2022-05-14:
> The most important part of the stack trace is not shown, we never encountered this issue, so not being able to reproduce it, you have to help us understand what's going on otherwise there's no way we can help.

I'm migrating from docker-openwisp as published from hub.docker.com to building my own images using the current master.

Here's my logs; please advise how to get more verbose stack trace if this is insufficent:

Waiting for database to become available...
Connection with database established.
Waiting for redis to become available...
Connection with redis established.
Operations to perform:
Apply all migrations: account, admin, auth, authtoken, check, config, connection, contenttypes, device_monitoring, firmware_upgrader, geo, monitoring, openwisp_ipam, openwisp_notifications, openwisp_radius, openwisp_users, pki, reversion, sessions, sites, socialaccount, subnet_division, topology
Running migrations:
Traceback (most recent call last):
File "/opt/openwisp/manage.py", line 9, in <module>
execute_from_command_line(sys.argv)
File "/usr/local/lib/python3.10/site-packages/django/core/management/__init__.py", line 446, in execute_from_command_line
utility.execute()
File "/usr/local/lib/python3.10/site-packages/django/core/management/__init__.py", line 440, in execute
self.fetch_command(subcommand).run_from_argv(self.argv)
File "/usr/local/lib/python3.10/site-packages/django/core/management/base.py", line 414, in run_from_argv
self.execute(*args, **cmd_options)
File "/usr/local/lib/python3.10/site-packages/django/core/management/base.py", line 460, in execute
output = self.handle(*args, **options)
File "/usr/local/lib/python3.10/site-packages/django/core/management/base.py", line 98, in wrapped
res = handle_func(*args, **kwargs)
File "/usr/local/lib/python3.10/site-packages/django/core/management/commands/migrate.py", line 290, in handle
post_migrate_state = executor.migrate(
File "/usr/local/lib/python3.10/site-packages/django/db/migrations/executor.py", line 131, in migrate
state = self._migrate_all_forwards(
File "/usr/local/lib/python3.10/site-packages/django/db/migrations/executor.py", line 163, in _migrate_all_forwards
state = self.apply_migration(
File "/usr/local/lib/python3.10/site-packages/django/db/migrations/executor.py", line 248, in apply_migration
state = migration.apply(state, schema_editor)
File "/usr/local/lib/python3.10/site-packages/django/db/migrations/migration.py", line 131, in apply
operation.database_forwards(
File "/usr/local/lib/python3.10/site-packages/django/db/migrations/operations/special.py", line 193, in database_forwards
self.code(from_state.apps, schema_editor)
File "/usr/local/lib/python3.10/site-packages/openwisp_monitoring/check/migrations/0003_create_ping.py", line 14, in create_device_ping
auto_create_ping(
File "/usr/local/lib/python3.10/site-packages/celery/local.py", line 188, in __call__
return self._get_current_object()(*a, **kw)
File "/usr/local/lib/python3.10/site-packages/celery/app/task.py", line 392, in __call__
return self.run(*args, **kwargs)
File "/usr/local/lib/python3.10/site-packages/openwisp_monitoring/check/tasks.py", line 63, in auto_create_ping
has_check = Check.objects.filter(
File "/usr/local/lib/python3.10/site-packages/django/db/models/manager.py", line 85, in manager_method
return getattr(self.get_queryset(), name)(*args, **kwargs)
File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 1071, in filter
return self._filter_or_exclude(False, args, kwargs)
File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 1089, in _filter_or_exclude
clone._filter_or_exclude_inplace(negate, args, kwargs)
File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 1096, in _filter_or_exclude_inplace
self._query.add_q(Q(*args, **kwargs))
File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1502, in add_q
clause, _ = self._add_q(q_object, self.used_aliases)
File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1532, in _add_q
child_clause, needed_inner = self.build_filter(
File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1377, in build_filter
lookups, parts, reffed_expression = self.solve_lookup_type(arg)
File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1187, in solve_lookup_type
_, field, _, lookup_parts = self.names_to_path(lookup_splitted, self.get_meta())
File "/usr/local/lib/python3.10/site-packages/django/db/models/sql/query.py", line 1677, in names_to_path
raise FieldError(
django.core.exceptions.FieldError: Cannot resolve keyword 'check_type' into field. Choices are: active, check, content_type, content_type_id, created, description, id, modified, name, object_id, params
Applying check.0003_create_ping...

nemesisdesign, 2022-05-15:
@kitchung it says that it can't find the check_type field on the model, but it's there in the updated code:
https://github.com/openwisp/openwisp-monitoring/blob/dedc4a913004e019a74eef6fcb64647ecdf31114/openwisp_monitoring/check/base/models.py#L34-L39

I would check `/usr/local/lib/python3.10/site-packages/openwisp_monitoring/check/base/models.py` to see if the file matches what's here on github.

Lugaresi, 2022-06-09:
I'm having the same issue on a fresh update to Debian Buster.
Upgrading was done via ansible and all packages involved are installed inside the virtualenv created during installation of OpenWISP.
Python is at version 3.9, tried deleting the virtualenv and the error persist.
I can confirm that models.py inside virtualenv matches the one posted here.
Bypassing this check just fails at the next one.

AndyX90, 2022-08-01:
Did someone solve the issue somehow? In my case it is an upgrade from an earlier version without monitoring enabled to a new version with monitoring enabled by default..

nemesisdesign, 2022-08-02:
Please try:

```
cd /opt/openwisp2/
source env/bin/activate
pip install -U --force-reinstall openwisp-monitoring "django~=4.0.0"
./manage.py migrate
```

And let us know how it goes.

Lugaresi, 2022-08-02:
Hello,
 thank you for trying to help us solve this issue.
I tried re-running ansible after cleaning up the playbook file, ansible-openwisp2 on this machine is at version 22.05.1.
After it failed in the usual spot I ran the commands you suggested. It still fails, and here's the full log of what happened.
[openwisp_manual_migrate.txt](https://github.com/openwisp/openwisp-monitoring/files/9241909/openwisp_manual_migrate.txt)

nemesisdesign, 2022-08-02:
Please try again with the following:

```
cd /opt/openwisp2/
source env/bin/activate
pip install -U --force-reinstall https://github.com/openwisp/openwisp-monitoring/tarball/fix-migrations
./manage.py migrate
```

And let us know how it goes.

AndyX90, 2022-08-03:
> Please try again with the following:
> 
> ```
> cd /opt/openwisp2/
> source env/bin/activate
> pip install -U --force-reinstall https://github.com/openwisp/openwisp-monitoring/tarball/fix-migrations
> ./manage.py migrate
> ```
> 
> And let us know how it goes.

@nemesisdesign After that, i re-run the playbook and everything is fine. Thank you very much!

nemesisdesign, 2022-08-04:
Anyone else can double check? @Lugaresi @kitchung @alangecker
I am sorry it took long but we were busy on other more urgent fronts.

Lugaresi, 2022-08-04:
Hello,
 thank you! Sorry it took a while, I can confirm that the steps work for me also, I followed the steps used by @AndyX90 and it succeeded both on manage.py and the playbook. The web interface is now operational and everything seems to be working fine.
 Thanks again for the support.

closed 2022-08-04 by nemesisdesign

nemesisdesign, 2022-08-04:
Thanks for confirming, I released a bug fix version: https://pypi.org/project/openwisp-monitoring/1.0.2/ .

Any affected user can fix this with:

```bash
cd /opt/openwisp2/
source env/bin/activate
pip install -U --force-reinstall openwisp-monitoring 
./manage.py migrate
```

commit e151a9c6 (2022-08-04): [fix] Fix creation of checks for existing devices #348

Fixes #348
PR #423 (2022-08-04): [fix] Fix creation of checks for existing devices #348
Fixes #348.

Checks:

- [x] I have manually tested the proposed changes
- N/A: I have written new test cases to avoid regressions (if necessary)
- N/A: I have updated the documentation (e.g. README.rst)
commit 148f4cf5 (2022-08-04): [fix] Fix creation of checks for existing devices #348

Fixes #348

(cherry picked from commit e151a9c6f7d7859881467a454bf4372e805912fc)