openwisp/openwisp-monitoring #349: [change] Remove charts which have empty data

nemesisdesign, 2022-03-21:
- [ ] In the API which returns the charts of a device, if the data is empty, there's no need to include the chart in the response.
- [ ] We shall make sure the chart is not shown in the UI

proshee, 2022-03-22:
when i run the command python manage.py migrate,
it shows
File "/Users/tanujchandela/Documents/git/Contribution/venv/openwisp-monitoring/tests/openwisp2/init.py", line 3, in <module>
from .celery import app as celery_app
File "/Users/tanujchandela/Documents/git/Contribution/venv/openwisp-monitoring/tests/openwisp2/celery.py", line 5, in <module>
from celery import Celery
ModuleNotFoundError: No module named 'celery'
but i have installed celery
(base) tanujchandela@Tanujs-MacBook-Air tests % pip show celery
Name: celery
Version: 5.2.3
Summary: Distributed Task Queue.
Home-page: http://celeryproject.org
Author: Ask Solem
Author-email: auvipy@gmail.com
License: BSD
Location: /Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages
Requires: billiard, click, click-didyoumean, click-plugins, click-repl, kombu, pytz, setuptools, vine
Required-by: openwisp-controller, openwisp-notifications
please help me with this issue

n1snt, 2022-04-02:
Can I take this issue?

PR #371 (2022-05-05): [change] Removed charts which have empty data #349
- If data in charts is empty, then we can remove it from `API` response `(api/v1/monitoring/device/)` and it no longer visible on UI.
- For charts having empty data, we can check value of its `traces`.

### Before (If charts have empty data) : 

[image]

[image]

### After (showing only those charts having data) :

[image]

Closes #349

Checks:

- [x] I have manually tested the proposed changes
- [ ] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)
commit 4dbed037 (2022-05-05): [change] Removed charts which have empty data #349

Closes #349
closed 2022-05-05 by nemesisdesign