openwisp/openwisp-monitoring #350: [feature] Add wifi version capability to wifi interface status

nemesisdesign, 2022-04-07:
In the device status, we're able to see different WiFi info like channel, frequency, noise, etc, but it's not evident whether that interface is working with in High Throughput mode (802.11n/wifi4), Very High Throughput mode (802.11ac/wifi5) or High Efficiency (802.11ax/wifi6).

It would be great to add this information.

Aryamanz29, 2023-01-11:
To add wifi version capability to wifi interface status, I think first we need to add the `wifi-device` -> [`htmode`](https://github.com/openwisp/openwrt-openwisp-monitoring/search?q=htmode) option to [`netjson-monitoring`](https://github.com/openwisp/openwrt-openwisp-monitoring/blob/7001e772395212e76c0279b53315a7bb634aedb5/openwisp-monitoring/files/sbin/netjson-monitoring.lua#L85-L94) and then update the code here.

Ref : https://openwrt.org/docs/guide-user/network/wifi/basic#wi-fi_devices

```diff
diff --git a/openwisp-monitoring/files/sbin/netjson-monitoring.lua b/openwisp-monitoring/files/sbin/netjson-monitoring.lua
index 7965769..6190746 100755
--- a/openwisp-monitoring/files/sbin/netjson-monitoring.lua
+++ b/openwisp-monitoring/files/sbin/netjson-monitoring.lua
@@ -90,7 +90,8 @@ for _, radio in pairs(wireless_status) do
           tx_power = iwinfo.txpower,
           signal = iwinfo.signal,
           noise = iwinfo.noise,
-          country = iwinfo.country
+          country = iwinfo.country,
+          htmode = radio.config.htmode
         }
       }
       if iwinfo.mode == 'Ad-Hoc' or iwinfo.mode == 'Mesh Point' then
```

nemesisdesign, 2023-01-11:
@Aryamanz29 yes but `htmode` should be contained in the `wireless` section.

commit bd7a684d (2023-01-13): [feature] Added wifi version capability to wifi interface status #350

Closes #350
commit 7965a5b8 (2023-01-20): [feature] Added wifi version capability to wifi interface status #350

Closes #350
PR #488 (2023-01-20): [feature] Added wifi version capability to wifi interface status #350
- **Depends on :** https://github.com/openwisp/openwrt-openwisp-monitoring/pull/118.

[image]

Closes #350

Checks:

- [x] I have manually tested the proposed changes
- [x] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)
commit 1db262b1 (2023-01-20): [feature] Added wifi version capability to wifi interface status #350

Closes #350
closed 2023-01-20 by nemesisdesign