openwisp/openwisp-monitoring #357: [bug] Chart > 1 month looking weird

nemesisdesign, 2022-04-19:
The bar chart with timeframe higher than one month is not looking right:

[image]

The bars became too thin.

nemesisdesign, 2022-05-24:
@Aryamanz29 yes the bars should look thicker, there shouldn't be so much space in between, very weird what's going on there, it wasn't happening before. I guess we will have to ask on the plotly support channels to see if they can hint at what's going on, if we can't find a solution.

nemesifier, 2024-04-01:
Today we did some debugging with @pandafy and we found out the issue is caused by daylight time saving changes, which make influxdb return weird date times in the response, eg:

```json
{
    "charts": [
        {
            "traces": [
                [
                    "wifi_clients",
                    [
                        0,
                        0,
                        0,
                        3,
                        1,
                        0,
                        0,
                        0,
                        0,
                        0,
                        1,
                        4,
                        3,
                        1,
                        2,
                        1,
                        1,
                        2,
                        4,
                        3,
                        2,
                        0,
                        0,
                        1,
                        2,
                        2,
                        1,
                        2,
                        0,
                        0,
                        0
                    ]
                ]
            ],
            "summary": {
                "wifi_clients": 4
            },
            "description": "Unique WiFi clients count of the entire network.",
            "title": "General WiFi Clients",
            "type": "bar",
            "unit": "",
            "summary_labels": [
                "Total Unique WiFi clients"
            ],
            "colors": [
                "#1f77b4"
            ],
            "colorscale": null
        },
        {
            "traces": [
                [
                    "download",
                    [
                        0.02,
                        0.02,
                        0.01,
                        1.38,
                        0.29,
                        null,
                        null,
                        null,
                        null,
                        null,
                        0.47,
                        3.47,
                        2.04,
                        5.85,
                        1.91,
                        0.08,
                        7.01,
                        2.76,
                        7.77,
                        6.59,
                        0.26,
                        0.05,
                        null,
                        0.34,
                        0.88,
                        4.99,
                        1.68,
                        2.11,
                        0.03,
                        0.03,
                        0.03,
                        0.02
                    ]
                ],
                [
                    "upload",
                    [
                        0.0074,
                        0.0072,
                        0.0064,
                        0.66,
                        0.0088,
                        null,
                        null,
                        null,
                        null,
                        null,
                        0.17,
                        0.45,
                        0.16,
                        0.17,
                        0.32,
                        0.02,
                        0.7,
                        2.13,
                        0.82,
                        3.17,
                        0.05,
                        0.04,
                        null,
                        0.06,
                        0.08,
                        0.57,
                        0.09,
                        0.08,
                        0.01,
                        0.01,
                        0.01,
                        0.0067
                    ]
                ]
            ],
            "summary": {
                "upload": 9.83,
                "download": 50.07
            },
            "description": "Network traffic of the whole network (total, download, upload).",
            "title": "General Traffic",
            "type": "stackedbar+lines",
            "unit": "adaptive_prefix+B",
            "summary_labels": [
                "Total traffic",
                "Total download traffic",
                "Total upload traffic"
            ],
            "colors": [
                "#7f7f7f",
                "#1f77b4",
                "#ff7f0e"
            ],
            "colorscale": null,
            "fill": "none",
            "trace_type": {
                "download": "stackedbar",
                "upload": "stackedbar",
                "total": "lines"
            },
            "trace_order": [
                "total",
                "download",
                "upload"
            ],
            "calculate_total": true
        }
    ],
    "x": [
        "2024-03-02 00:00",
        "2024-03-03 00:00",
        "2024-03-04 00:00",
        "2024-03-05 00:00",
        "2024-03-06 00:00",
        "2024-03-07 00:00",
        "2024-03-08 00:00",
        "2024-03-09 00:00",
        "2024-03-10 00:00",
        "2024-03-11 00:00",
        "2024-03-12 00:00",
        "2024-03-13 00:00",
        "2024-03-14 00:00",
        "2024-03-15 00:00",
        "2024-03-16 00:00",
        "2024-03-17 00:00",
        "2024-03-18 00:00",
        "2024-03-19 00:00",
        "2024-03-20 00:00",
        "2024-03-21 00:00",
        "2024-03-22 00:00",
        "2024-03-23 00:00",
        "2024-03-23 23:00",
        "2024-03-24 00:00",
        "2024-03-25 00:00",
        "2024-03-26 00:00",
        "2024-03-27 00:00",
        "2024-03-28 00:00",
        "2024-03-30 00:00",
        "2024-03-31 00:00",
        "2024-04-01 00:00"
    ]
}
```

In the response above, March 23 is repeated twice, while March 29 is missing.
Our hypothesis is that this kind of result confuses the library and breaks its automatic width calculation.

AnirudhSinghRajora, 2026-10-27:
Hey, looks like this issue has been stale for a while. I would like to work on this!