openwisp/openwisp-monitoring #383: [bug] Title of chart does not update

pandafy, 2022-05-30:
The `wifi_clients` metric does not have data for in 1 day time period, but have data in 1 month time period. When I switch between different time periods, the WiFi Client charts is displayed/hidden respectively (this is an expected feature). 

But, when a new chart is added, the title of the old chart is not updated. In the video below, you can see that the WiFi Client chart have "br-lan" label. This is a **bug**. 

https://user-images.githubusercontent.com/32094433/171042501-f1dd324c-5cfb-466c-ac1b-9c2163ac2e34.mp4

**Expected outcome**
The title of the chart should get updated when any chart is added or removed from the webpage.

Aryamanz29, 2023-04-28:
I tried to reproduce this bug with the latest master, but it seems to be working as expected. The chart title updates when I switch between different time periods. In my case, I have WiFi client data available for the last 30 days but not for the last 1-day time period (please see below).

https://user-images.githubusercontent.com/56113566/235131379-f237ff98-d432-4e78-b491-6da3860e7851.mp4

pandafy, 2026-10-21:
Closing as solved!

closed 2026-10-21 by pandafy