openwisp/openwisp-monitoring #390: [change] Configure celery beat for Iperf check

Aryamanz29, 2022-06-02:
- We need to configure celery beat for Iperf check so that it can be run at a scheduled time.
```py
# Ex.
# Task schedule for 5 AM and 12 PM
@periodic_task(run_every=crontab(minute=0, hour='5,12'))
```

pandafy, 2022-06-03:
> * Using celery [crontab schedules](https://docs.celeryq.dev/en/latest/userguide/periodic-tasks.html#crontab-schedules) we can configure check task to run on specific time, It also covers most of the time variations for scheduling celery tasks.

The celery beat is configured at the project level in settings.py. We only need to configure the celery beat in the `tests/openwisp2/settings.py` for this issue. Later, we will port this change to `ansible-openiwsp2` and `docker-openwisp`. 

https://github.com/openwisp/openwisp-monitoring/blob/872f7db21f36325ed417307e0fec9446ed6d59d7/tests/openwisp2/settings.py#L170-L177

**Don't hard code the time in the celery task**

okraits, 2022-06-03:
> The celery beat is configured at the project level in settings.py. We only need to configure the celery beat in the `tests/openwisp2/settings.py` for this issue. Later, we will port this change to `ansible-openiwsp2` and `docker-openwisp`.

People who don't use `ansible-openwisp2` or `docker-openwisp` can still configure celery-beat in their `settings.py`, right?

pandafy, 2022-06-04:
> People who don't use ansible-openwisp2 or docker-openwisp can still configure celery-beat in their settings.py, right?

Yes. It is just another Django settings. 😄

nemesisdesign, 2022-06-08:
Let's keep in mind that by default we should recommend to run this at night or during non peak traffic times to not interfere with work or standard traffic.

nemesisdesign, 2022-07-07:
Ok so we need to be able to pass a list of checks to exclude and include right? So we can have the 5 minutes check perform anything except iperf and we can configure another entry only for iperf which runs at other intervals.

commit b7b7795b (2022-07-08): [docs] Added Iperf check celery beat configuration #390

Closes #390
commit cbb93f79 (2022-07-08): [change] Configured celery beat for Iperf check in settings #390

Closes #390
commit dc0316a9 (2022-08-09): [feature] Iperf #385 #390 #398 #399 #406 #405 #412 #414 #417 #418 #422

Closes #385
Closes #390
Closes #398
Closes #399
Closes #405
Closes #412
Closes #406
Closes #414
Closes #417
Closes #418
Closes #422
commit 17cb5167 (2022-08-09): [feature] Iperf #385 #390 #398 #399 #406 #405 #412 #414 #417 #418 #422

Closes #385
Closes #390
Closes #398
Closes #399
Closes #405
Closes #412
Closes #406
Closes #414
Closes #417
Closes #418
Closes #422
closed 2022-10-20 by nemesisdesign