openwisp/openwisp-monitoring #406: [feature] Add possibility to add alerts for related_fields

Aryamanz29, 2022-06-24:
- If we look at current implementation of alerts, We see that alerts checks thresholds on only `value/field_name`, it would be nice if we can add alert functionality for `extra_values/related_fields` in as well.
https://github.com/openwisp/openwisp-monitoring/blob/6119ac0dfa7221e7e06d88fdcd628ec0d629a3a7/openwisp_monitoring/monitoring/base/models.py#L315-L358

nemesisdesign, 2022-06-25:
> * If we look at current implementation of alerts, We see that alerts checks thresholds on only `value/field_name`, it would be nice if we can add alert functionality for `extra_values/related_fields` as well.
>   https://github.com/openwisp/openwisp-monitoring/blob/6119ac0dfa7221e7e06d88fdcd628ec0d629a3a7/openwisp_monitoring/monitoring/base/models.py#L315-L358

We need this for the Iperf alerts right? If so, we can look for a solution, do you have any idea?

Aryamanz29, 2022-06-26:
> We need this for the Iperf alerts right? If so, we can look for a solution, do you have any idea?
Yes, It would be great If we can add alerts on `related_fields` because my `iperf` configuration currently look like this :
https://github.com/openwisp/openwisp-monitoring/blob/issue-385/iperf-check/openwisp_monitoring/monitoring/configuration.py#L546-L650

commit db55c684 (2022-07-14): [feature] Added alert for related_field (Iperf jitter) #406

Closes #406
commit dc0316a9 (2022-08-09): [feature] Iperf #385 #390 #398 #399 #406 #405 #412 #414 #417 #418 #422

Closes #385
Closes #390
Closes #398
Closes #399
Closes #405
Closes #412
Closes #406
Closes #414
Closes #417
Closes #418
Closes #422
commit 8e60b808 (2022-08-09): [feature] Added alert_on_related_fields #406

Closes #406
commit 34c2dbc5 (2022-08-09): [feature] Added alert_on_related_fields #406

Closes #406
closed 2022-10-20 by nemesisdesign