openwisp/openwisp-monitoring #414: [feature] Add support for iperf3 authentication

Aryamanz29, 2022-07-15:
- With this enhancement, the user can use **RSA authentication** between the **client** and the **server** to restrict the connections to the server **only to legitimate clients.**

- We can leverage `OPENWISP_IPERF_CHECK_CONFIG` for this enhancement.

- The following **options** of `iperf3` are can be used to achieve this:
   #### **At server side**
   -  `--rsa-private-key-path` :   path to the RSA private key used to decrypt authentication credentials.
   - ` --authorized-users-path` :   path to the configuration file containing user credentials (`.csv (format: username,sha256)`)
   
   ####  **At client side**
   -  `--username` :   username for authentication.
   - ` --rsa-private-key-path` : path to the RSA public key used to encrypt authentication credentials.
   -  The password will be prompted for interactively when the test is run. 
   - **Note:** the password to use can also be specified via the **IPERF3_PASSWORD** environment variable. If this variable is present, the password prompt will be skipped.
  
## Steps

### Server

1. Generate RSA keypair.
```bash
openssl genrsa -des3 -out private.pem 2048
openssl rsa -in private.pem -outform PEM -pubout -out public.pem
openssl rsa -in private.pem -out private_not_protected.pem -outform PEM
```
After these commands, the **public key** will be contained in the file **`public.pem`** (will be used with **--rsa-private-key-path**  at client) and the **private key** will be contained in the file **`private_not_protected.pem.`** (will be used with **--rsa-private-key-path**  at server)

2. Then create a `credentials.csv` file with a hashed password.
```bash
S_USER=mario S_PASSWD=rossi
echo -n "{$S_USER}$S_PASSWD" | sha256sum | awk '{ print $1 }'
----
0b0c98028105e9e4d3f100280eac29bba90af614d1c75612729228e4d160c601 #This is the hash of "mario"
```
`credentials.csv` 
```
# file format: username,sha256
mario, bf7a49a846d44b454a5d11e7acfaf13d138bbe0b7483aa3e050879700572709b
```
3. Now start the server with auth options.
```bash
iperf3 -s --rsa-private-key-path ./private_not_protected.pem --authorized-users-path ./credentials.csv
```
### Client

1. Copy **`public.pem`** from server to client (public key is used to encrypt the authentication token containing the user credentials)
```bash
# scp username@IP:source_path destination_path
scp aryaman@192.168.5.109:Desktop/public.pem .
```
2. Now run iperf3 test in client mode with auth options and enter user password.
```bash
iperf3 -c 192.168.5.109 --rsa-public-key-path public.pem --username mario
```
3. At server side you will see
```bash

```

nemesisdesign, 2022-07-15:
@Aryamanz29 great work! In the README we can explain that on the OpenWrt side the iperf3-ssl package is needed for this feature, in prod we will mostl ikely use RSA authentication which is better.

Aryamanz29, 2022-07-18:
> @Aryamanz29 great work! In the README we can explain that on the OpenWrt side the iperf3-ssl package is needed for this feature, in prod we will mostl ikely use RSA authentication which is better.

Hey @nemesisdesign we need to provide both `--rsa-private-key-path` `--authorized-users-path` (at server side) & `--username` `--rsa-private-key-path` (at client side) for this feature.

```bash
# server
iperf3: parameter error - you must specify a path to a valid RSA private key and
# client
iperf3: parameter error - you must specify a username, password, and path to a valid RSA public key
```

commit dc0316a9 (2022-08-09): [feature] Iperf #385 #390 #398 #399 #406 #405 #412 #414 #417 #418 #422

Closes #385
Closes #390
Closes #398
Closes #399
Closes #405
Closes #412
Closes #406
Closes #414
Closes #417
Closes #418
Closes #422
commit 17cb5167 (2022-08-09): [feature] Iperf #385 #390 #398 #399 #406 #405 #412 #414 #417 #418 #422

Closes #385
Closes #390
Closes #398
Closes #399
Closes #405
Closes #412
Closes #406
Closes #414
Closes #417
Closes #418
Closes #422
commit 465d5f5f (2022-08-26): [feature] Iperf #385 #390 #398 #399 #406 #405 #412 #414 #417 #418 #422

Closes #385
Closes #390
Closes #398
Closes #399
Closes #405
Closes #412
Closes #406
Closes #414
Closes #417
Closes #418
Closes #422
commit 073e1ba1 (2022-09-07): [feature] Iperf #385 #390 #398 #399 #406 #405 #412 #414 #417 #418 #422

Closes #385
Closes #390
Closes #398
Closes #399
Closes #405
Closes #412
Closes #406
Closes #414
Closes #417
Closes #418
Closes #422