openwisp/openwisp-monitoring #415: [bug] Traffic total sometimes is very wrong

nemesisdesign, 2022-07-15:
This is happening in the current development version (1.1.0 alpha).

[image]

The total amount in some cases does not match the download + upload exactly, but it should.

pandafy, 2022-07-18:
I think this is happening because of the approximations. `0.0015`, got rounded off to `0.002`

pandafy, 2022-08-12:
This issue occurs because we round-off the values of the trace individually using the following method: 

https://github.com/openwisp/openwisp-monitoring/blob/e563520c7765d4e638f8e65b8b8e4095fa2fc4cc/openwisp_monitoring/monitoring/base/models.py#L631-L639

**Note:** The `decimal_places` default to `2`. 

Breaking down the data points from the chart screenshot given in issue description. The `rx_bytes` is `0.0005GB` and the `tx_bytes` is `0.01` GB. Note that these values are obtained after rounding-off the initial values. 

The total bytes that we have in this case is `0.0105GB` which is greater than the control value `0.01`. Hence, the approximation will be done for two decimal places, i.e. `0.01GB`. This results in statcked-bars overshooting the total trace line.

Here's another example: 

[image]

After rounding off, we have following values in the graph
```
rx_bytes: 20MB
x_bytes: 5.5MB
total_bytes: 30MB
```

We get 30MB here because the approximations are done in GB in the backend. 

```
rx_bytes: 0.02GB
x_bytes: 0.0055GB
total_bytes: 0.030GB
```

Here, the total bytes will be `0.0255GB` which is greater than the control value `0.01`, hence it get rounded off to `0.03` (two decimal places. 

A solution to avoid these problems would be to calculate the `total_bytes` by adding `rx_bytes` and `tx_bytes` after rounding them.

nemesisdesign, 2022-08-12:
It seems the easiest solution may be to avoid rounding if the chart has the adaptive byte filter configured on it, so we let the JS in the frontend do the rounding.

commit e81289ea (2022-08-17): [fix] Fixed traffic charts showing wrong traffic total #415

The traffic charts were showing the wrong total because of the
approximation (round-off) done in the backend to decrease the
the precision of values to two decimal places.

Since the adaptive feature already does this approximation
in the frontend, the rounding off is skipped from the backend.

Closes #415
commit 694ff82b (2022-08-18): [fix] Fixed traffic charts showing wrong traffic total #415

The traffic charts were showing the wrong total because of the
approximation (round-off) done in the backend to decrease the
the precision of values to two decimal places.

Since the adaptive feature already does this approximation
in the frontend, the rounding off is skipped from the backend.

Closes #415
PR #430 (2022-08-31): [fix] Fixed traffic charts showing wrong traffic total #415
The traffic charts were showing the wrong total because of the
approximation (round-off) done in the backend to decrease the
the precision of values to two decimal places.

Since the adaptive feature already does this approximation
in the frontend, the rounding off is skipped from the backend.

Closes #415

Checks:

- [x] I have manually tested the proposed changes
- [x] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)
commit 4ef6fba9 (2022-08-31): [fix] Fixed traffic charts showing wrong traffic total #415

The traffic charts were showing the wrong total because of the
approximation (round-off) done in the backend to decrease the
the precision of values to two decimal places.

After attempting to fix this in the backend we concluded that it
would have been best fixed with JS code in the frontend.
The total amount of traffic is now calculated entirely
at client side level.

Closes #415
closed 2022-08-31 by nemesisdesign