openwisp/openwisp-monitoring #439: [bug][iperf] Iperf check raises error if setting is not configured

pandafy, 2022-09-22:
Iperf check raises `KeyError` if setting is not configured for an organization. We should handle this exception and log at ERROR level that Iperf settings are not configuration for the organization.

Aryamanz29, 2022-09-23:
@pandafy I think we've already covered Iperf check invalid configuration in [`Iperf check on multiple servers`](https://github.com/openwisp/openwisp-monitoring/pull/434) PR. https://github.com/openwisp/openwisp-monitoring/blob/6933988dbad756da2dfcc7bbdd1e7f8c47e4c364/openwisp_monitoring/check/classes/iperf.py#L123-L132

####  `Tests`
https://github.com/openwisp/openwisp-monitoring/blob/6933988dbad756da2dfcc7bbdd1e7f8c47e4c364/openwisp_monitoring/check/tests/test_iperf.py#L639-L665

Aryamanz29, 2022-09-23:
@pandafy I have checked this scenario with https://github.com/openwisp/openwisp-monitoring/pull/434 PR and I think it's working as expected. BTW, I made some [minor improvements](https://github.com/openwisp/openwisp-monitoring/pull/434/commits/6933988dbad756da2dfcc7bbdd1e7f8c47e4c364) in the PR in order to work like this.

#### Device registered with org `Default`:
##### `Iperf configuration`
```py
OPENWISP_MONITORING_IPERF_CHECK_CONFIG = {
    # org : default
    '1b4b50b1-7950-4735-8d4d-424ed98df1a4': {
        # available iperf servers
        'host': ['192.168.1.10'],
        'username': 'arya',
        'password': 'arya@1234',
        'rsa_public_key': (
            """
            MIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAwuEm+iYrfSWJOupy6X3N
            dxZvUCxvmoL3uoGAs0O0Y32unUQrwcTIxudy38JSuCccD+k2Rf8S4WuZSiTxaoea
            6Du99YQGVZeY67uJ21SWFqWU+w6ONUj3TrNNWoICN7BXGLE2BbSBz9YaXefE3aqw
            GhEjQz364Itwm425vHn2MntSp0weWb4hUCjQUyyooRXPrFUGBOuY+VvAvMyAG4Uk
            msapnWnBSxXt7Tbb++A5XbOMdM2mwNYDEtkD5ksC/x3EVBrI9FvENsH9+u/8J9Mf
            2oPl4MnlCMY86MQypkeUn7eVWfDnseNky7TyC0/IgCXve/iaydCCFdkjyo1MTAA4
            BQIDAQAB
            """
        ),
        'client_options': {
            'port': 5201,
            # 'parallel': 2,
            #'reverse': True,
            #'bidirectional': True,
            'connect_timeout': 1,
            # 'blockcount': '1K',
            # 'bytes': '100M',
            'udp': {'bitrate': '30M', 'length': '1460'},
            'tcp': {'bitrate': '0', 'length': '256K'},
        },
    }
}
```

#### Device org changed to `another-org` and the config remains the same:

[image]

[image]

pandafy, 2022-09-23:
Thanks @Aryamanz29! I have added this issue in the PR description of #439. So, this gets closed when that PR is merged.

Aryamanz29, 2022-10-20:
Solved and Closed by : https://github.com/openwisp/openwisp-monitoring/pull/447

closed 2022-10-20 by Aryamanz29