openwisp/openwisp-monitoring #453: [docs] Mention custom date range usage in `List of endpoints` section

Aryamanz29, 2022-10-26:
We recently added a date range widget to the project, which updates existing API endpoints (General & Device charts), so we need to update the [List of endpoints](https://github.com/openwisp/openwisp-monitoring#list-of-endpoints) section of the docs.

kishan3, 2022-10-27:
Can you tell what to add in the doc? I will be working on it. Thanks

Aryamanz29, 2022-10-30:
Hey, @kishan3 Thanks for showing interest in this issue but this one is connected to the [feature](https://github.com/openwisp/openwisp-monitoring/commit/c6700bc76cd2aed1ab5d112bb4e02fe802248b41) that we recently introduced to our project. If you're looking for some `hactoberfest/good-first-issues`[ here is the list](https://github.com/openwisp/openwisp-utils/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc+label%3AHacktoberfest) 😄

commit 9d1b463b (2022-10-27): [docs] Mentioned usage of custom daterange #453

Closes #453
commit 362ff9ed (2022-10-27): [docs] Mentioned usage of custom daterange #453

Closes #453
commit 9e5b75cb (2022-10-27): [docs] Mentioned usage of custom daterange #453

Closes #453
commit b4fa0112 (2022-12-11): [docs] Mentioned usage of custom daterange #453

Closes #453
closed 2022-12-13 by nemesisdesign