openwisp/openwisp-monitoring #474: [change] Add timeout to all celery tasks

nemesisdesign, 2022-12-27:
Let's ensure all tasks have a reasonable timeout.

Some tasks like `run_checks` must have a higher timeout (eg: 1 hour), we should avoid hitting this under all conditions, all the other tasks should have much shorter timeout but not too short.

uzaxirr, 2023-01-22:
Hey, Can i take this up?
My approach would be adding `CELERYD_TASK_SOFT_TIME_LIMIT` configuration option in celery configuration file. However can you please specify what should be the timeout value

pandafy, 2023-01-23:
Hey @uzaxirr! This is a delicate change which requires prior experience with OpenWISP. I ask you to check "good first issues" on [OpenWISP Contributors Board]((https://github.com/orgs/openwisp/projects/3).

PR #496 (2023-01-30): [change] Added default timeout to celery tasks #474
Closes #474

Checks:

- [ ] I have manually tested the proposed changes
- [ ] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)
commit 79f3e627 (2023-01-30): [change] Added default timeout to celery tasks #474

Closes #474
closed 2023-01-30 by nemesisdesign