openwisp/openwisp-monitoring #559: Server Error page when creating new metric

Darin755, 2024-01-30:
I am not sure if this is the proper place to report this but I don't understand the OpenWisp internals so I figured this is a good place to start. I am trying to setup monitoring of client devices but when I go to create the new metric I just get the following page:
[image]

I don't know where to start troubleshooting so I'm posting this here in order to see if I can get some help. My setup is Openwisp ansible in a LXC container behind a reverse proxy and Netbird for the VPN connection. Netbird is just wireguard so that shouldn't be an issue. Additionally I have already installed the Openwisp monitoring package on all of the client devices.

Does anyone know where to start looking for errors?

nemesifier, 2024-02-23:
@Darin755 check the logs in /opt/openwisp2.
Can you also share the procedure you followed to replicate the issue?

closed 2024-02-23 by nemesifier

Darin755, 2024-02-25:
Thanks for your time