openwisp/openwisp-monitoring #562: [feature] Add a URL in the device's change page to view the device on the full map

pandafy, 2024-02-06:
Depends on https://github.com/openwisp/openwisp-monitoring/issues/561

We need to add an URL to the device page that redirects the user to the dedicated maps view (added in above issue). 

When the dedicated map view is opened, the map should automatically adjust the zoom level and center the map according to the desired device.

pandafy, 2024-02-06:
Open for discussion: 

We can add a "View on map" link in the object tools like this

[image]

dee077, 2025-05-15:
## Approach 
- We can override the **change_list.html** template from openwisp-controller to add a **VIEW ON MAP** button. This button will link to a newly created admin view and template dedicated to displaying the device location on a map.

pandafy, 2025-05-16:
@dee077 ensure that this button is only shown when the device has a DeviceLocation object.

commit cf9511c3 (2025-09-17): [feature] Setting url fragments and applying to that data #562

Set URL fragments when clicking nodes in geo and indoor maps, and apply state from URL on load.

Fixes #562