openwisp/openwisp-monitoring #564: [feature] Allow to drill down to indoor map (floor plan) from map

pandafy, 2024-02-06:
We need to allow viewing indoor map (floorplan) for locations directly from the map. We should also show all the floors available for that location, and show floorplan for the selected floor.

pandafy, 2024-02-09:
These are a rough mock-ups I created

[image]

[image]

dee077, 2025-05-14:
## Approach
- Add a **View on Floorplan** button to the device card for a specific location.
- Clicking the button will display a floorplan map instance on a separate layer above the existing map container. A prototype has already been made, [deployed](https://openwisp-multimap-prototype.vercel.app/), and the source code is available on my [github](http://github.com/dee077/Openwisp-Multimap-Prototype).
- Take inspiration from Google Maps’ multi-floor display to switch between floors.
- Implement a UI to toggle between different floor levels within the same map view.

### Here is the sample Desktop view:

[image]

A responsive design similar to this should be created for mobile and tablet views.

commit 8e8bf373 (2025-07-26): [feature] Add floorplan map above dashboard map #564

- Added a Floorplan button below the device table

- On click a new container is created and renders the floorplan map using Netjsongraph

- Currently using static data from Netjsongraph's examples

- Will integrate dynamic data fetching via REST API and render on AJAX success

Fixes #564
commit 366febcf (2025-09-10): [feature] Add floorplan map above dashboard map #564

- Added a Floorplan button below the device table

- On click a new container is created and renders the floorplan map using Netjsongraph

- Currently using static data from Netjsongraph's examples

- Will integrate dynamic data fetching via REST API and render on AJAX success

Fixes #564

[feature] Integrate floorplan coordinates endpoint from controller

[feature] Add multiple floorplan navigation
commit 7c44bdc7 (2025-09-17): [feature] Added indoor maps #564 #662

- Allow to drill down to indoor map from geographic map
- Added REST API endpoints for getting device details and indoor coordinates
  for specific locations
- Allow searching and filtering by status in geographic map locations

Closes #564
Closes #662

---------