openwisp/openwisp-monitoring #572: Status info not sent if IMEI missing

mips171, 2024-04-30:
Possible bug when monitoring modemmanager interfaces - When the IMEI is not found, because it is a required value, all attempts to send the rest of the monitoring data will fail. The IMEI may not be readable at all times because ModemManager may be restarting, or the modem may be rebooting or crashed, or simply unused.

Also sorry, I had the exact error message logged in response.txt but my terminal decided to delete that, so I don't have it anymore. But it said something to the effect of "imei is required".

Opening it here rather than with the agent because the error looked to me like it came from the API.

```shell
Wed May  1 08:17:46 2024 daemon.err openwisp-monitoring[11960]: > POST //api/v1/monitoring/device/[redacted]/?key=[redacted]&time=23-01-2024_03:27:51.000000 HTTP/1.1
Wed May  1 08:17:46 2024 daemon.err openwisp-monitoring[11960]: > Host: mgmt.networkhoist.com
Wed May  1 08:17:46 2024 daemon.err openwisp-monitoring[11960]: > User-Agent: curl/7.83.1
Wed May  1 08:17:46 2024 daemon.err openwisp-monitoring[11960]: > Accept: */*
Wed May  1 08:17:46 2024 daemon.err openwisp-monitoring[11960]: > Content-Type: application/json
Wed May  1 08:17:46 2024 daemon.err openwisp-monitoring[11960]: > Content-Length: 13723
Wed May  1 08:17:46 2024 daemon.err openwisp-monitoring[11960]: >
Wed May  1 08:17:46 2024 daemon.err openwisp-monitoring[11960]: } [13723 bytes data]
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < HTTP/1.1 400 Bad Request
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Server: nginx
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Date: Tue, 30 Apr 2024 22:17:47 GMT
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Content-Type: application/json
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Content-Length: 93
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Connection: keep-alive
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Vary: Accept, Accept-Language, Cookie
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Allow: GET, POST, HEAD, OPTIONS
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < X-Frame-Options: DENY
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Content-Language: en-gb
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < X-Content-Type-Options: nosniff
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Referrer-Policy: same-origin
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < X-XSS-Protection: 1; mode=block
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < X-Content-Type-Options: nosniff
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Referrer-Policy: same-site
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Content-Security-Policy: default-src http: https: data: blob: 'unsafe-inline'; script-src 'unsafe-eval' https: 'unsafe-inline' 'self'; frame-ancestors 'self'; connect-src https://mgmt.networkhoist.com wss: 'self'; worker-src https://mgmt.networkhoist.com blob: 'self';
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Permissions-Policy: interest-cohort=()
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: < Strict-Transport-Security: max-age=31536000
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: <
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: { [93 bytes data]
Wed May  1 08:17:47 2024 daemon.warn root: Data not sent successfully. Retrying in 32 seconds
Wed May  1 08:17:47 2024 daemon.err openwisp-monitoring[11960]: root: Data not sent successfully. Retrying in 32 seconds
```

mips171, 2024-05-01:
@nemesifier If there's no SIM inserted, ModemManager will put the modem into a `failed` state, and you lose a lot of basic functionality and information. A workaround may be to use `qmicli` (`qmicli -p -d /dev/cdc-wdm0 --dms-get-ids`), or `mbimcli` to query the modem directly. A couple drawbacks with that approach are 1) those utilities need to be installed (they're not included by default), 2) there's no structured output (json or key/value) so you have to parse raw output with regular expressions. There's a small risk that can break with each update.

Would it be possible to make IMEI optional?

nemesifier, 2025-04-19:
@ghost sorry for not replying on this, I was very busy with other aspects of OpenWISP.
I suppose anything is possible, but we need to understand what use case we're trying to address. Does it really make sense to have a mobile interface without an IMEI? In what situations would that be legitimate? I'm asking genuinely, as I'm not an expert in this area.

If the mobile interface is expected to be a cellular modem, then missing an IMEI might indicate a problem (e.g., misconfigured modem, emulated device, or even faulty hardware). In that case, it could be valuable for OpenWISP Monitoring to detect and flag the absence.