openwisp/openwisp-monitoring #629: [feature] Monitor and display WiFi session hostnames

c-gabri, 2025-01-19:
**Is your feature request related to a problem? Please describe.**
It is difficult to identify WiFi clients only by their MAC addresses, especially when there are many of them.

**Describe the solution you'd like**
If the hostname associated to a MAC address is know to an AP (e.g. because it runs a DHCP server or has its `/etc/ethers` regularly populated by a DHCP server), the AP should send this information to the OpenWisp server, which should display it as an additional field in the corresponding WiFi session.

dee077, 2025-03-03:
In the wifi session, there is already a `DEVICE` column. However, the reason devices are displayed as MAC addresses instead of hostnames is that, in the Device section, the hostname is set to the `MAC address`. This results in the MAC address being shown instead of hostname.

In the below image it is shown the device name is set to `Test-Device1`

- The fact that all of the wifi sessions are shown below has same Device name is because it is been populated by a manual api which uses the `Test-Device1`'s `pk` and `key` in the params 

[image]

#### This is what I interpreted, please correct me if I'm wrong.

c-gabri, 2025-03-04:
@dee077 In the `DEVICE` column I see the hostnames of the OpenWisp APs, but this feature request is about the hostnames _of the wifi clients connected to them_.

dee077, 2025-03-04:
#### Thanks for the clarification!

KaizokuHam, 2026-10-02:
Hi, I'd like to investigate this issue.

From the discussion, I understand that the existing Device column refers to the managed access point, whereas the requested hostname belongs to each connected WiFi client.

I'm planning to trace how WiFi session data flows from the OpenWrt monitoring agent to the API and dashboard.

Would the expected approach be to include an optional hostname in the WiFi client data, using DHCP lease information where available, while keeping the MAC address as a fallback?

Is this feature still planned?

nemesifier, 2026-10-02:
@KaizokuHam this feature likely requires changing the agent and the server. It's not suited for beginner contributors.

kiwimato, 2026-10-29:
@nemesifier I needed this for my own network, so I've implemented both sides on my forks. It has been running since early October on three OpenWrt 25.12.5 APs (gen-1 UniFi, big-endian MIPS) managed by OpenWISp.

Server (openwisp-monitoring): WifiSession gets hostname and ip_address fields, filled from the DHCP leases the agent already sends. IPv4 is preferred when a client has several addresses. A * hostname counts as unknown, and a known value is never overwritten with a blank one. The new fields show up in the admin list, detail view and search, and in the REST API serializer. The branch also includes a migration, docs and tests.
[actual changes for openwisp-monitoring](https://github.com/openwisp/openwisp-monitoring/compare/master...kiwimato:openwisp-monitoring:issues/629-wifi-session-hostname)

Agent (openwrt-openwisp-monitoring): this part is for APs that don't run the DHCP server themselves. It adds an optional openwisp-monitoring-dhcp-snooping package (ucode) that snoops DHCP traffic and merges the leases it sees into the DHCP data the agent reports. Without the package, the agent behaves exactly as it does today.
[actual changes for openwrt-openwisp-monitoring/](https://github.com/openwisp/openwrt-openwisp-monitoring/compare/master...kiwimato:openwrt-openwisp-monitoring:dhcp-snooping)

On the server side the existing tests and QA checks pass. The only failures were the Selenium map tests, and those fail the same way on unmodified master in my environment. The agent package builds with the 25.12.5 SDK and installs on a clean x86-64 rootfs as well as on the MIPS APs.

Would you like PRs for these? If so, should I open a companion issue in openwrt-openwisp-monitoring for the agent side first?