openwisp/openwisp-monitoring #658: [feature] Improve Celery logging to include check name for long-running tasks

pandafy, 2025-05-07:
**Is your feature request related to a problem? Please describe.**

Currently, when Celery tasks in openwisp-monitoring take longer to execute, it's difficult to identify which specific check is being executed. This makes it hard to debug or monitor performance issues effectively.

**Describe the solution you'd like**
Enhance the Celery logging output to include the check.name (or other identifying information) in the log line, especially when tasks are being executed. This would allow operators to immediately see which checks are running and help diagnose slow tasks.

pandafy, 2025-05-08:
We can add debug logs in the check class for the time taken by `check` and `store` methods.

nemesifier, 2025-05-10:
> We can add debug logs in the check class for the time taken by `check` and `store` methods.

Sounds good!

commit d7216f7e (2025-05-13): [enhancement] Log execution time of monitoring checks #658

Closes #658
commit 78d1a79b (2025-05-13): [enhancement] Log execution time of monitoring checks #658

Closes #658
commit b8cfc376 (2025-05-14): [enhancement] Log execution time of monitoring checks #658

Closes #658
PR #659 (2025-05-15): [feature] Log execution time of monitoring checks #658

## Checklist

- [x] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [x] I have manually tested the changes proposed in this pull request.
- [x] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- [ ] I have updated the documentation.

## Reference to Existing Issue

Closes #658
closed 2025-05-15 by nemesifier